jim-easterbrook/pywws #30: Using EmonCms

bigbig6, 2015-08-26:
Hello,

Is it possible to add a service for pulling data into a EmonCms server?

Thank's.

Regards,

Big

jim-easterbrook, 2015-08-26:
If the server accepts HTML post or get (data in the URL) requests then it should be easy enough to add.

bigbig6, 2016-06-03:
Hello,

sorry for the long time...

Yes, this server accept HTML post and get:
http://IP/emoncms/input/post.json?node=VALUE1&time=NNNNNNNN&json={power:150,light:78}&apikey=APIKEY

IP is the IP address of the server, node is a value (a parameter that don't change) APIKEY is a number defined, power, light are names of column we need to feed.
We can have much more of this, probably one for each value of PYWWS. The value after is the one to feed in.
NNNNNNNN is the unix epoch or a formatted UTC time.

Regards

Big

jim-easterbrook, 2016-06-04:
That looks simple enough apart from the wrapping of two data items in a "json={...}" string. You need to create two files in `src/pywws/services`: `emoncms.ini` and `emoncms_template_1080.txt` (or `emoncms_template_3080.txt` if your station is a 3080 type). Then you need to use setup.py to build and install in the usual way.

Your `emoncms.ini` file would be something like this:
```
[config]
url = http://IP/emoncms/input/post.jsong
catchup = 0
use get = False
result = []
auth_type	= None

[fixed]
apikey = *APIKEY
node = *VALUE1
```
The `apikey` and `node` values would then be set in `weather.ini`. The `result` value may need setting, depending on what your server returns under normal conditions.

With just one data item your `emoncms_template_1080.txt` file would look something like this:
```
#live#
{
#idx "'time' : '%s',"#
#calc "expression_to_compute_power" "'json' : '{power:%.0f}',"#
}
```
If you need two values in your `json` value you probably need to make the expression return a tuple of two values and use expand the `'{power:%.0f}'` expression to use both values. I've not tried anything like this so you'll be charting new territory.

bigbig6, 2016-06-10:
Thank's a lot for your answer.
I'll get a try.

Regards

closed 2018-04-12 by jim-easterbrook