jim-easterbrook/pywws #33: Filter for data spikes

ThomasRgbg, 2015-11-21:
Hi, 

I have from time to time the problem that the data connection from the outside sensor seems to be not reliable and so I get corrupted data. But always just one sample. 

Of course this is a hardware issue and should primary addressed on this level. 

But would it be possible to add to pywws a filter, which could drop samples ? 

Like, if one sample is N digits different than the one before/afterwards, drop it. 

Thomas

jim-easterbrook, 2015-11-21:
There is an example "user calibration" module to do this for temperature spikes with a median filter. This is perhaps a bit over the top and a simple "is this value a lot more/less than the previous one" would probably be good enough, depending on your particular hardware and its problems. Fine Offset have used different RF modulation schemes over the years and they have different interference susceptibilities, so there isn't a one size fits all solution.
https://github.com/jim-easterbrook/pywws/blob/master/src/pywws/examples/modules/remove_temperature_spikes.py

ThomasRgbg, 2015-11-21:
This sounds pretty suitable. Is this calib function always called when data is imported ?

jim-easterbrook, 2015-11-22:
Yes, as part of the "processing" step 'raw' data is converted to 'calib' data by the user calibration module. When your templates select 'raw' data they actually get the 'calib' data.
http://pywws.readthedocs.org/en/latest/api/pywws.calib.html

closed 2018-04-12 by jim-easterbrook