jim-easterbrook/pywws #35: running pywws.Hourly fails with python error

sdineley, 2016-02-05:
My pywws installation has stopped getting data from my Maplin USB weather station. The python error output is:
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/pywws/Hourly.py", line 57, in <module>
    from . import Process
  File "/usr/local/lib/python2.7/dist-packages/pywws/Process.py", line 90, in <module>
    from .TimeZone import STDOFFSET
  File "/usr/local/lib/python2.7/dist-packages/pywws/TimeZone.py", line 41, in <module>
    import pytz
ValueError: bad marshal data (unknown type code)
```

I'm unsure if this is a corruption with the data my weather station has recorded, or a problem with pywws. Is there any additional debugging or log files that would be useful?

jim-easterbrook, 2016-02-05:
Nothing in that stacktrace suggests it's corruption in pywws's data. Try just importing pytz into python:
`python -c 'import pytz'`
I suspect reinstalling or upgrading (or downgrading) pytz may help.

sdineley, 2016-02-05:
Thanks for the quick reply!
`sudo easy_install-2.7 --upgrade pytz`
did the trick :+1:

closed 2016-02-05 by sdineley