jim-easterbrook/pywws #36: MQTT not working in Python3

matt2005, 2016-02-17:
I've build a new Raspberry Pi with Debian Jessie Lite, I've copied the data and configs from the Wheezy install. Installed pip and run sudo pip install paho-mqtt.
Setup pywws via pip and it uploads to MetOffice and Weather Underground, however when i add MQTT to livelog it crashes. This previously worked fine before i rebuilt the RPi, I suspect it's due to Python3.

Below is the error message
sudo python3 -m pywws.LiveLog -v /home/pi/weather
21:32:13:pywws.Logger:pywws version 15.12.0
21:32:13:pywws.Logger:Python version 3.4.2 (default, Oct 19 2014, 13:31:11)
[GCC 4.9.1]
21:32:13:pywws.WeatherStation.CUSBDrive:using pywws.device_libusb1
21:32:15:pywws.Calib:Using default calibration
21:32:38:pywws.weather_station:setting sensor clock 38.2162
21:33:13:pywws.LiveLog:unorderable types: str() <= int()
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/pywws/LiveLog.py", line 88, in LiveLog
    tasks.do_live(data)
  File "/usr/local/lib/python3.4/dist-packages/pywws/Tasks.py", line 300, in do_live
    self._do_common(['live'], calib_data)
  File "/usr/local/lib/python3.4/dist-packages/pywws/Tasks.py", line 269, in _do_common
    self._do_queued_tasks()
  File "/usr/local/lib/python3.4/dist-packages/pywws/Tasks.py", line 174, in _do_queued_tasks
    elif service.send_data(timestamp, prepared_data):
  File "/usr/local/lib/python3.4/dist-packages/pywws/toservice.py", line 339, in mqtt_send_data
    mosquitto_client.connect(hostname, port)
  File "/usr/local/lib/python3.4/dist-packages/paho/mqtt/client.py", line 611, in connect
    self.connect_async(host, port, keepalive, bind_address)
  File "/usr/local/lib/python3.4/dist-packages/paho/mqtt/client.py", line 667, in connect_async
    if port <= 0:
TypeError: unorderable types: str() <= int()
Exception ignored in: <bound method status.__del__ of <pywws.DataStore.status object at 0xb63da410>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/pywws/DataStore.py", line 109, in __del__
    self.flush()
  File "/usr/local/lib/python3.4/dist-packages/pywws/DataStore.py", line 117, in flush
    self._config.write(of)
  File "/usr/lib/python3.4/configparser.py", line 895, in write
    for section in self._sections:
  File "/usr/lib/python3.4/collections/__init__.py", line 87, in __iter__
    curr = root.next
ReferenceError: weakly-referenced object no longer exists

jim-easterbrook, 2016-02-18:
The `__del__` related stuff is just because it failed during initialisation. The important bit is
```
File "/usr/local/lib/python3.4/dist-packages/paho/mqtt/client.py", line 611, in connect
self.connect_async(host, port, keepalive, bind_address)
File "/usr/local/lib/python3.4/dist-packages/paho/mqtt/client.py", line 667, in connect_async
if port <= 0:
TypeError: unorderable types: str() <= int()
```
It appears that pywws is passing a string `port` value where an int is expected. The port value is taken from your config (as far as I can tell, I don't use the mqtt stuff).

matt2005, 2016-02-18:
I thought that but the config file hasn't changed since it was working on Wheezy.
Here is the mqtt section from the config. Can we force the port value to be read as a int as it seems that it may be reading as a string in Python3.

[mqtt]
topic = weatherstation
hostname = localhost
port = 1883
client_id = pywws
retain = true
auth = false
template = default
user = unknown
password = unknown

jim-easterbrook, 2016-02-18:
All that appears to be needed is to change line 316 of toservice.py to cast the string to an int, as long as that's compatible with earlier versions of paho-mqtt (I assume upgrading paho-mqtt is the cause of your problem - you could try installing an earlier version to test that).
https://github.com/jim-easterbrook/pywws/blob/master/src/pywws/toservice.py#L316

matt2005, 2016-02-18:
Thanks for the help, i've fixed it in my version and submitted a pull request

jim-easterbrook, 2016-02-18:
Have you tested it with earlier versions of paho-mqtt? I'd like to know if this is why it stopped working or whether pywws will need to specify a minimum version of it after merging your PR.

matt2005, 2016-02-18:
Both my installs have paho-mqtt 1.1, the only difference is the python versions. Wheezy python 2.7 and Jessie Python 3.4. The patch works on Jessie, and doesn't break the Wheezy version.

closed 2016-02-19 by jim-easterbrook