jim-easterbrook/pywws #38: USBError: [Errno 110] Operation timed out

wuppi83, 2016-03-09:
Hi Jim,
hello together,

First  ... much respect for your work !! Thank you !!!

At the moment i have a problem. after a few minutes i get the USBError: [Errno 110] Operation timed out. Does somebody have an idea what i can do ?!

Thanks

Andreas

jim-easterbrook, 2016-03-09:
You haven't said what you're running it on, but I assume a Raspberry Pi. The Pi's USB implementation has been buggy in the past, so make sure you've updated the firmware and operating system. Another thing you can try is putting a USB hub between the Pi and the weather station, or trying a different hub if you already have one.

wuppi83, 2016-03-09:
Thank you for the fast answer. Yes .. i use an raspberry. And all the things you recommend i have checked :0( Same Error.

wuppi83, 2016-03-09:
Is there an command to write only one livelog in the livedata.log .. not as a loop

jim-easterbrook, 2016-03-09:
The pywws-hourly command will fetch all data since it was last run, process it and then exit. Despite its name it can be run at 15 minute intervals (if you have a 5 minute logging interval).

mbirth, 2016-09-02:
By the way: This only seems to apply to the original Raspberry Pi. I have the RasPi 2 and never had any problems with USB devices at all. My dad has the RasPi 3 with the WH1080 connected to it and there also were no problems for now.

limpkin, 2016-12-18:
Hey everyone,

On a completely separate project running a RPi3 I also encountered this problem. In my case, the solution was to explicitly specify the timeouts when reading or writing, particularly for the latter: self.epout.write(arraytosend, 10000)

Regards

jim-easterbrook, 2018-06-12:
I recently bought a Pi Zero W and ran into this problem. After some investigation I discovered that the time outs only happen after about five seconds or more USB inactivity, i.e. when pywws "pauses" as it knows there won't be any new data for another 48 seconds.

I surmised that the Zero's OTG port was temporarily switching from "host" to "peripheral" mode. I cured the problem by editing `/boot/config.txt` and adding `dtoverlay=dwc2,dr_mode=host` at the end. There's more info about these "overlays" in `/boot/overlays/README`.

closed 2019-04-16 by jim-easterbrook