jim-easterbrook/pywws #40: Allow start and stop to take the current year and month as parameters

PaulCheffus, 2016-07-26:
Hi

I am finding that showing the last 12 months means that the graph moves across the screen as the month progresses when I only use <duration>weeks = 52</duration>

Would it be possible in the plot to use the following syntax for start and stop
<stop>year=%Y, month=%m, day =31</stop>
Which would allow you to specify the stop date as the end of the current month. Obviously the number of days would need to be calculated but 31 could be used to indicate the end of the month.

Then the start date time would be 52 weeks prior which should stop the graph moving from right to left as the days in the month increase. It would only then move as the month changes.

The image below shows the issue
[image]

Cheers

Paul

jim-easterbrook, 2016-07-26:
You can already set some elements of the <start> and <stop> values without introducing a new syntax.
http://pywws.readthedocs.io/en/latest/api/pywws.Plot.html#start
You could use this to set <stop> to day 28 of the current month, which would be safe.

PaulCheffus, 2016-07-28:
Hi

I am currently trialing this but so far it hasn't made any difference.

Cheers

Paul

closed 2017-01-11 by jim-easterbrook