jim-easterbrook/pywws #41: Twitter UTF-8 character count

mbirth, 2016-08-30:
I've seen in the code that ToTwitter.py limits the length of a tweet to 140 characters without attached media and 117 characters with media attached.

To calculate the length, the generated tweet is converted to utf-8 first. And then, the first 117 (or 140) characters are used.

However, this counts multi-byte characters as multiple characters (`😎` = 4 characters, although it should be counted as only 1 character) where the [Twitter doc](https://dev.twitter.com/overview/api/counting-characters) clearly states that multi-byte characters count as one character.

jim-easterbrook, 2016-08-31:
The last time I did some tests (a few weeks ago) posting messages with any non-ASCII characters in reduced the maximum length of the message, regardless of what Twitter's API docs say. The python-twitter library has it's own idea of how to limit the length, and gets it very wrong in version 3. (But it does allow the checking to be disabled in version 3.)
I'll do some more testing.

mbirth, 2016-08-31:
[This snippet](https://gist.github.com/mtigas/810514) from @mtigas might be of interest. Twitter also uses Unicode Normalization Form C.

Current method:

```
>>> status = u"09:59 CEST: 🌡24,7℃, 🌀  SSW, Ø 2 ㎞/h, 🔝 6 ㎞/h, Luftfeuchte: 41%, 💧 (stdl.) 0,0㎜, Druck: 1023 h㎩, langsam fallend"
>>> len(status)
111
>>> len(status.encode('utf8'))
134
>>> status = status.encode('utf8')[:117].decode('utf8', errors='ignore')
>>> status
u'09:59 CEST: \U0001f32124,7\u2103, \U0001f300  SSW, \xd8 2 \u339e/h, \U0001f51d 6 \u339e/h, Luftfeuchte: 41%, \U0001f4a7 (stdl.) 0,0\u339c, Druck: 1023 h\u33a9'
```

As you can see, it cut everything from the last comma.

NFC:

```
>>> from unicodedata import normalize
>>> len(status)
111
>>> len(normalize("NFC", status))
111
>>> status = normalize("NFC", status)
>>> status
u'09:59 CEST: \U0001f32124,7\u2103, \U0001f300  SSW, \xd8 2 \u339e/h, \U0001f51d 6 \u339e/h, Luftfeuchte: 41%, \U0001f4a7 (stdl.) 0,0\u339c, Druck: 1023 h\u33a9, langsam fallend'
```

Looking good.

jim-easterbrook, 2016-08-31:
Normalising malformed Unicode is a separate issue, and shouldn't be needed if your templates are OK. The problem is how different versions of each Python twitter library interpret the 140 character limit. If they wrongly assume it's bytes then that's what pywws has to accommodate by truncating the string more than is strictly necessary.

jim-easterbrook, 2016-08-31:
I don't know what was going wrong in my earlier tests, but I'm now able to post 140 character strings with non-ASCII characters in. Commit 3919ae2 should fix the problem.

mbirth, 2016-08-31:
python-twitter just checks the `len(status)` and even recognises URLs and counts them as 23 characters. So everything is fine from the libary's point of view.

And why does pyWWS have to do a pre-check when the Twitter library checks the length again?

If you insist on getting the tweet out (instead of throwing an error message), a probably better way would be to directly use twitter_utils.calc_expected_status_length() from python-twitter and shorten according to that.

mbirth, 2016-08-31:
Awesome! Thank you! :+1:

jim-easterbrook, 2016-08-31:
Python-twitter (in at least one version) considers decimal numbers to be URLs. This is why I disabled its check on the length. (It rejects over long tweets rather than truncating them.)
I think it is sensible to truncate a weather tweet - most people would put the important information first any way.

mbirth, 2016-09-01:
And tested it:

[image]

140 characters incl. multi-byte characters. Perfect!

closed 2016-09-01 by mbirth

Peeklearee, 2017-11-25:
My weather station Twitter Account has been updated to support 280 characters (Hoorah). 

I see that https://dev.twitter.com/overview/api/counting-characters hasn’t been updated to support this change of message length. I’ll ask Twitter about that.

Does pywws need to change too before I can include more info in my hourly tweet? 

Do I need to change any config to take affect this change?

jim-easterbrook, 2017-11-25:
Version 17.11.0 of pywws allows 280 characters if you are using the python-twitter library. I don't know the situation with Tweepy. The change log is here: https://github.com/jim-easterbrook/pywws/releases
You'll probably want to edit your Twitter template to take advantage of the increased length.

Peeklearee, 2017-11-26:
Jim, sorry, never saw the update notes on GitHub. Updated pywws to 17.11.0 and works as documented. Thanks.