jim-easterbrook/pywws #42: Please add support for candlesticks graphs

mbirth, 2016-09-06:
A candlesticks graph with whiskers (and/or whiskerbars) would be ideal for those day/night avg/max/min graphs currently used in 28days and yearly graphs.

Those look like these: http://gnuplot.sourceforge.net/demo/candlesticks.html . Or this one:

[image]

Sadly, the current source code doesn't allow this as candlesticks graphs need 4 y-values per row (NightAvg, NightMin, DayMax, DayAvg) and the Plot.py only writes 1 values per row.

I'm not sure but it might be sufficient if the `<ycalc>` field would allow a tuple to specify multiple data values. And the `using 1:2` string for gnuplot should be adapted accordingly.

jim-easterbrook, 2016-09-07:
See CONTRIBUTING.rst for instructions on how to add a new feature and create a pull request.  (-:

mbirth, 2016-09-09:
A little teaser. Pull request coming soon...

[image]

closed 2016-09-11 by jim-easterbrook