jim-easterbrook/pywws #49: pywws-testweatherstation got error

YimengZhu, 2017-06-28:
Hi,

Thanks a lot for your work in this module. Unfortunately I didn't success to run  pywws-Westweatherstation

I just followed the documentation of pywws and run it on a Mac. The weather station I'm using now is WH1080. After I installed the dependencies and pywws(with pip, in python 3) I tried to start pywws-testweatherstation, then I got the following error:

16:49:28:pywws.Logger:pywws version 16.12.0, build 1367 (e917ba9)
Traceback (most recent call last):
  File "/Users/yimeng/anaconda/bin/pywws-testweatherstation", line 11, in <module>
    sys.exit(main())
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/TestWeatherStation.py", line 117, in main
    raw_fixed = ws.get_raw_fixed_block()
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/WeatherStation.py", line 598, in get_raw_fixed_block
    self._fixed_block = self._read_fixed_block()
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/WeatherStation.py", line 642, in _read_fixed_block
    result += self._read_block(mempos)
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/WeatherStation.py", line 630, in _read_block
    new_block = self.cusb.read_block(ptr)
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/WeatherStation.py", line 265, in read_block
    if not self.dev.write_data(buf):
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/device_ctypes_hidapi.py", line 145, in write_data
    if hidapi.hid_write(self.device, ctypes.c_char_p(data), size) != size:
TypeError: bytes or integer address expected instead of str instance

Then I followed [this link](https://stackoverflow.com/questions/37888565/python-3-5-ctypes-typeerror-bytes-or-integer-address-expected-instead-of-str) that I changed 
`if hidapi.hid_write(self.device, ctypes.c_char_p(data), size) != size:`
in device_ctypes_hidapi.py line 145 to 
`if hidapi.hid_write(self.device, ctypes.c_char_p(data.encode('UTF-8')), size) != size:`
and solved this exception.

After that I ran pywws-testweatherstation again and I got the following error:
16:54:43:pywws.Logger:pywws version 16.12.0, build 1367 (e917ba9)
Traceback (most recent call last):
  File "/Users/yimeng/anaconda/bin/pywws-testweatherstation", line 11, in <module>
    sys.exit(main())
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/TestWeatherStation.py", line 117, in main
    raw_fixed = ws.get_raw_fixed_block()
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/WeatherStation.py", line 598, in get_raw_fixed_block
    self._fixed_block = self._read_fixed_block()
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/WeatherStation.py", line 642, in _read_fixed_block
    result += self._read_block(mempos)
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/WeatherStation.py", line 630, in _read_block
    new_block = self.cusb.read_block(ptr)
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/WeatherStation.py", line 267, in read_block
    return self.dev.read_data(32)
  File "/Users/yimeng/anaconda/lib/python3.5/site-packages/pywws/device_ctypes_hidapi.py", line 125, in read_data
    'pywws.device_ctypes_hidapi.USBDevice.read_data failed')
OSError: pywws.device_ctypes_hidapi.USBDevice.read_data failed

I checked  n = hidapi.hid_read_timeout(self.device, data, length, 100) in device_ctypes_hidapi.py line line 122 and found n = 0 at this place, which means no data to be read before time out. Besides the self.device is no problem which is passed a integer to the method. 

Did you meet the same problem before? For any help I will be appreciate!

Thanks in advance

jim-easterbrook, 2017-06-28:
I've no experience with Mac OS (I stopped using Macs when MacOS 8.6 was current). As you've found, the hid_api library does not give any helpful error message. Does it behave any differently if you run pywws-testweatherstation as the root user?

YimengZhu, 2017-06-28:
Unfortunately no difference. I don't think it is because of the sudo. Is there any walk around off the hidapi?

YimengZhu, 2017-06-28:
Furthermore I'm using python 3.5. Is any problems with the python version?

jim-easterbrook, 2017-06-28:
As I understand it, MacOS forces the weather station to be an HID device, so the hidapi library has to be used. The pywws installation process uses 2to3 to generate Python3 code from the Python2 source - I'm not aware of any problems with this, but it might be worth trying Python2.

YimengZhu, 2017-06-29:
Hi, 

I changed to ubuntu and it works!

But I got another problem with mqtt again. According to the documentation I configured the weather.ini file like following 
`
[config]
ws type = 1080
usb activity margin = 3.0
pressure offset = 27.7
asynchronous = True
frequent writes = False
gnuplot version = 4.2
gnuplot encoding = iso_8859_1
day end hour = 21
rain day threshold = 0.2

[paths]
work = /tmp/weather
templates = /home/yimeng/weather/templates/
graph_templates = /home/yimeng/weather/graph_templates/
local_files = /home/yimeng/weather/results/

[ftp]
local site = False
site = ftp.username.your_isp.co.uk
user = username
password = secret
directory = public_html/weather/data/
secure = False
port = 21

[live]
services = ['mqtt']
plot = []
text = []

[logged]
services = ['mqtt']
text = []
plot = []

[hourly]
services = []
text = []
plot = []

[12 hourly]
services = []
text = []
plot = []

[daily]
services = []
text = []
plot = []

[mqtt]
topic = /weather/pywws
hostname = localhost
port = 1883
client_id = pywws
retain = True
auth = False
user = unknown
password = unknown
template = default
`
Then I ran
`python -m pywws.toservice -vvv ~/Desktop/weather/data/ mqtt` 
and
` sudo pywws-livelog -vvv ~/Desktop/weather/data`
but I found in the pywws log that there is no sevice for mqtt. Did I miss something to make pywws publish to mqtt? Btw, the mosquitto is no problem, I can pub or sub with other client.

Thanks in advance

jim-easterbrook, 2017-06-29:
Were there no error messages (or any other messages) from `python -m pywws.toservice -vvv ~/Desktop/weather/data/ mqtt`?

YimengZhu, 2017-06-29:
No... The output is as following:
19:50:10:pywws.Logger:pywws version 16.12.0, build 1367 (e917ba9)
19:50:10:pywws.Logger:Python version 2.7.12 (default, Nov 19 2016, 06:48:10) 
[GCC 5.4.0 20160609]

Is this the right way to start those services? I'm wondering if I understood the doc correctly.

jim-easterbrook, 2017-06-29:
I've just tried it myself - that's all the output you get if you haven't configured a 'service' correctly. For most services you need to set a station ID and password or similar information in your weather.ini file.

YimengZhu, 2017-06-29:
Do you mean the [mqtt] section in weather.ini ?

Can you please post the correct output of a well configured service?

Thanks a lot.

jim-easterbrook, 2017-06-29:
Sorry, I don't know about mqtt. I think you should ask on the pywws mailing list - there are other mqtt users there.

YimengZhu, 2017-06-29:
Ok. Thanks.

closed 2017-10-12 by jim-easterbrook

mrumpf, 2020-12-30:
I got this error on a clean Raspberry Pi 4 installation and I could resolve it by issuing the command "apt install python3-libusb1".