jim-easterbrook/pywws #52: ValueError: time data '2017@SynoEAStream' does not match format '%Y-%m-%d.txt'

vlczak, 2017-10-13:
Hello,
I run WH1080 on Synology NAS.
I did full reinstall of Synology, copy-pasted pywws data and installed pywws.

`python -m pywws.LogData -vvv /volume1/weather/`
returns this error:
`22:59:12:pywws.Logger:pywws version 17.01.0, build 1372 (7d691f1)
22:59:12:pywws.Logger:Python version 2.7.12 (default, Jun 26 2017, 02:36:18) 
[GCC 4.9.3 20150311 (prerelease)]
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/lib/python2.7/site-packages/pywws-17.1.0-py2.7.egg/pywws/LogData.py", line 309, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/site-packages/pywws-17.1.0-py2.7.egg/pywws/LogData.py", line 306, in main
    DataStore.data_store(root_dir)).log_data(sync=sync, clear=clear)
  File "/usr/lib/python2.7/site-packages/pywws-17.1.0-py2.7.egg/pywws/DataStore.py", line 507, in __init__
    core_store.__init__(self, os.path.join(root_dir, 'raw'))
  File "/usr/lib/python2.7/site-packages/pywws-17.1.0-py2.7.egg/pywws/DataStore.py", line 246, in __init__
    safestrptime(file, "%Y-%m-%d.txt").date())
  File "/usr/lib/python2.7/site-packages/pywws-17.1.0-py2.7.egg/pywws/DataStore.py", line 87, in safestrptime
    return datetime(*(time.strptime(date_string, format)[0:6]))
  File "/usr/lib/python2.7/_strptime.py", line 478, in _strptime_time
    return _strptime(data_string, format)[0]
  File "/usr/lib/python2.7/_strptime.py", line 332, in _strptime
    (data_string, format))
ValueError: time data '2017@SynoEAStream' does not match format '%Y-%m-%d.txt'`

Any idea what to fix to get a proper time?
Thanks a lot

jim-easterbrook, 2017-10-13:
One of your data directories (raw, calib, hourly etc) contains a file called `2017@SynoEAStream` which should not be there. If you can identify the file then give it the correct name, otherwise remove it.

vlczak, 2017-10-13:
Thanks Jim,
Locating folders and deleting helped. Issue solved.

https://mikebeach.org/2012/12/12/getting-rid-of-the-eadir-folders-on-synology-nas-dsm/

closed 2017-10-13 by vlczak