jim-easterbrook/pywws #57: Mosquitto - add SSL/TLS support

SBado, 2017-11-20:
Please, add SSL/TLS support for MQTT service.
I configured my broker to use a certificate from Let's Encrypt. To establish the connection, I just had to add this line to `toservice.py` (just before the call to `mosquitto_client.connect(hostname, int(port))`):

`mosquitto_client.tls_set("/etc/ssl/certs/DST_Root_CA_X3.pem")`

For this to work, there should be the possibility to configure the paths for `ca_certs`, `certfile` and `keyfile` in `weather.ini` (in my case, I just had to set set the `ca_certs` path).

GeekyTim, 2018-08-20:
Thanks :-) That's what I was planning to do, know I know which version to use.  I'll send a pull request once it's ready.

GeekyTim, 2018-10-11:
@jim-easterbrook O think this can be closed.

closed 2018-10-11 by jim-easterbrook