jim-easterbrook/pywws #58: can't compare datetime.datetime to NoneType when submitting to wunderground

bobintetley, 2018-04-02:
$ pywws-livelog -vvv ~/weather/data
16:56:14:pywws.Logger:pywws version 17.11.0, build 1380 (b01d94a)
16:56:14:pywws.Logger:Python version 2.7.3 (default, Nov 24 2017, 21:13:24) 
[GCC 4.6.3]
16:56:14:pywws.WeatherStation.CUSBDrive:using pywws.device_pyusb
16:56:15:pywws.weather_station:avoid 0.769716978073
16:56:17:pywws.Calib:Using default calibration
Traceback (most recent call last):
  File "/usr/local/bin/pywws-livelog", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/pywws/LiveLog.py", line 123, in main
    return LiveLog(args[0])
  File "/usr/local/lib/python2.7/dist-packages/pywws/LiveLog.py", line 76, in LiveLog
    asynch=asynch)
  File "/usr/local/lib/python2.7/dist-packages/pywws/Tasks.py", line 113, in __init__
    service_name=name)
  File "/usr/local/lib/python2.7/dist-packages/pywws/toservice.py", line 284, in __init__
    self.next_update = min(self.next_update, self.data.before(datetime.max))
TypeError: can't compare datetime.datetime to NoneType

self.data.before seems to be evaluating to None

bobintetley, 2018-04-02:
Temporarily fixed it in my version by adding a conditional:

        if self.data.before(datetime.max) is not None:
            self.next_update = min(self.next_update, self.data.before(datetime.max))

jim-easterbrook, 2018-04-03:
If `self.data.before(datetime.max)` is `None` then you have no data.

bobintetley, 2018-04-03:
I'm really new to pywws, but I'd followed the instructions and run the log process first - there was a month of catchup data from my station in ~/weather/data before I started trying to use live logging. Did I misunderstand and I'm missing something else? Everything is working great now, thanks for responding.

jim-easterbrook, 2018-04-03:
Hmm, something strange is happening then. (Or happened that once.)

jim-easterbrook, 2018-04-03:
I think what might have happened is that you started live logging without "processing" first (converting raw data to calibrated, hourly, daily, etc.). Live logging does the processing as soon as it gets a 'logged' data record, but may attempt to do stuff (such as uploading to Weather Underground) as soon as it gets a 'live' record. Most users get live logging running first before adding services such as Weather Underground, which is why I hadn't seen this before.

jim-easterbrook, 2018-04-03:
Commit c51fd36 should ensure this doesn't catch anyone else out in future.

bobintetley, 2018-04-03:
Ah, that's exactly what happened - I didn't bother processing the logs as I only wanted to update some online services. Thanks for coming up with a fix so quickly.

closed 2018-04-05 by jim-easterbrook