jim-easterbrook/pywws #64: Exception when running livelog

the-louie, 2018-06-03:
I've reinstalled pywws on a new Raspberry pi and tried to configure livelog. When i started i got this:

```
pywws-livelog -vvv ~/weather/data
10:26:22:pywws.Logger:pywws version 18.04.1, build 1389 (5a32227)
10:26:22:pywws.Logger:Python version 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516]
10:26:22:pywws.WeatherStation.CUSBDrive:using pywws.device_pyusb1
10:26:24:pywws.Calib:Using default calibration
Traceback (most recent call last):
  File "/usr/local/bin/pywws-livelog", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/pywws/LiveLog.py", line 126, in main
    return LiveLog(args[0])
  File "/usr/local/lib/python2.7/dist-packages/pywws/LiveLog.py", line 76, in LiveLog
    asynch=asynch)
  File "/usr/local/lib/python2.7/dist-packages/pywws/Tasks.py", line 113, in __init__
    service_name=name)
  File "/usr/local/lib/python2.7/dist-packages/pywws/toservice.py", line 284, in __init__
    self.next_update = min(self.next_update, self.data.before(datetime.max))
TypeError: can't compare datetime.datetime to NoneType
```

I did some debug-printing of the content of row 284 and found the following:
self.next_update = 2018-05-27 08:26:24.550716
datetime.max = 9999-12-31 23:59:59.999999
self.data.before = None

Created a quick workaround and got it running:
```
        if self.data.before(datetime.max) is not None:
            self.next_update = min(self.next_update, self.data.before(datetime.max))
```

jim-easterbrook, 2018-06-03:
You're getting this error because you have no data, so `self.data.before(datetime.max)` returns `None`. In normal setting up pywws.logdata will have been run before attempting live logging, so this error doesn't usually happen.

The functionality of pywws.toservice has been replaced by the modules in pywws.service, so it's going to be dropped before long.

the-louie, 2018-06-03:
Ok, thanks!

closed 2019-04-16 by jim-easterbrook