jim-easterbrook/pywws #69: MQTT Template in weather.ini being ignored

GeekyTim, 2018-08-29:
I have updated the MQTT template in weathwer.ini, and restarted pywws, but it is still sending out the old default template defined in mqtt.py.  For example, I don't have the temp_out_f, but it is still being received on the broker and subscribers.

jim-easterbrook, 2018-08-29:
Is this after upgrading to commit 4ca8d5b or later? Earlier versions were ignoring template manipulation done after initialising the base class. Running at very high verbosity (run the class directly with 'python -m pywws.service.mqtt -vvvv data_dir') should print out the template.

GeekyTim, 2018-08-29:
I'll upgrade again :-)

jim-easterbrook, 2018-08-29:
If you haven't already done so, read this before running after upgrading.
https://groups.google.com/d/msg/pywws/5p6tP8Z9r0s/dLaisMGZAgAJ

GeekyTim, 2018-08-29:
The update worked!  Thanks.

jim-easterbrook, 2018-08-29:
Thanks for volunteering to test all this for me.  (-:

GeekyTim, 2018-08-29:
No problem :-)

GeekyTim, 2018-09-06:
multi-topic = True
retain = False

jim-easterbrook, 2018-09-06:
No quotation marks or anything?

GeekyTim, 2018-09-06:
No. Does it need them?

jim-easterbrook, 2018-09-06:
No. But if they were there it explain your problem.

jim-easterbrook, 2018-09-06:
As you have `multi_topic = True` I would expect you to see multi-topic messages when you monitor it. It doesn't stop the usual single topic message.

The `retain` value is passed to the mosquitto.Client session, so if it's being ignored that sounds like a bug in paho.mqtt.client.

GeekyTim, 2018-09-06:
Ah! Misunderstood the multi-topic config.  That's working fine then :-)

retain does seem to be being ignored.  I'll look further.

closed 2018-10-11 by GeekyTim