jim-easterbrook/pywws #72: sudo pywws-testweatherstation -d -h 1 doesn't show some data

grossik, 2018-09-19:
I use testweatherstation -d -h 1 for my weatherstation and send this data in json to my FTP server. My problem is that I do not see some data like dewpoint. And rain data are not displayed correctly.
Sorry for my bad English.

jim-easterbrook, 2018-09-19:
Dewpoint is not included in the raw data read from the station, you need to calculate it from temperature and humidity.

The raw rain data is the total amount since the outside sensors were reset. To get the amount of rain in the last hour you need to subtract the data from an hour earlier.

grossik, 2018-09-19:
> Dewpoint is not included in the raw data read from the station, you need to calculate it from temperature and humidity.
> 
> The raw rain data is the total amount since the outside sensors were reset. To get the amount of rain in the last hour you need to subtract the value from an hour earlier.

Thank you. So I have to remember value of rain from the previous hour and deduct from new value?