jim-easterbrook/pywws #82: Live wind data to Weathercloud service

craigusus, 2019-05-08:
Hi,
Would it be possible that 'Live' wind data could be sent to Weathercloud?
From what I can see in [weathercloud.py](https://github.com/jim-easterbrook/pywws/blob/master/src/pywws/service/weathercloud.py) only 'wspdavg', 'wspdhi', and 'wdiravg' data is sent. Could 'wspd' and 'wdir' be added to weathercloud.py?

ie [PiWeatherUK](https://app.weathercloud.net/d0214731119#wind) only shows Average wind speed, direction and the gust. The live info is ghosted out as it is not being received.

Thanks,

jim-easterbrook, 2019-05-08:
The values sent are from the 'live' data. I only have a copy of v0.6 of their API, which is extremely brief and doesn't define the difference between `wspd` and `wspdavg`.

atrophicshiner, 2019-05-08:
My understanding of their API was the wind averages for weathercloud were meant to be over 10 minute intervals, the pywws module should really be sending the data as the current/live values. This would then mean the average graphs would be ghosted.

jim-easterbrook, 2019-05-08:
OK, that seems reasonable. I'll change it.

craigusus, 2019-05-08:
Thanks for the replies...
This was the response from Weathercloud:

> Hi Craig,
Thank you for using Weathercloud.
The wspdavg and wdiravg are the 10-min average wind speed and direction values.
Jim can calculate these 2 parameters or send the wspd and wdir values also as wspdavg and wdiravg.
He will find all the information in the API Documentation that we sent him.
Regards,
--
The Weathercloud team

So would Weathercloud calculate the 10 average from the uploaded 'live' data or would pywws have to do this before uploading?

jim-easterbrook, 2019-05-08:
Data is only uploaded to Weathercloud every 10 minutes (limit set in their API). If using logged data pywws would only have two values (for most users) with which to calculate an average. It's not really practical to provide 10-minute averages of any data.

craigusus, 2019-05-08:
@jim-easterbrook Thanks for the replies and the explanations.

closed 2019-11-28 by jim-easterbrook