jim-easterbrook/pywws #84: Unable to get Wind Chill in MQTT

GeekyTim, 2019-06-01:
I've tried formatting the wind_chill in the MQTT template, but for the life of me I cannot get it to work.  I get:

File "<unknown>", line 12
Jun 01 19:44:34 PiWeather pywws-livelog[1570]:     '#calc \'wind_chill(data('temp_out'], data['wind_ave'])\' \'"chill": "%.0f",\' \'\' \'scale(x, 10.0)\'#\n'

Do you have any idea what is happening here?

jim-easterbrook, 2019-06-02:
The template parser can't parse that expression, probably because you're using single quotes ``'`` for both the element names (e.g. ``'temp_out'``) and the "words" that make up the ``#calc#`` expression, (e.g. ``'wind_chill(data('temp_out'], data['wind_ave'])'``).

GeekyTim, 2019-06-02:
I have it outputting some data now.  Not sure how accurate it is at the moment, but I think I have got it now.

closed 2019-06-02 by GeekyTim

vlczak, 2019-06-06:
@GeekyTim 
I noticed you had `data('temp_out']` instead of `data['temp_out']`
Could you post here your final working template for others?

Thanks

jim-easterbrook, 2019-06-06:
That's a mistake I hadn't spotted. Should definitely be square brackets.

GeekyTim, 2019-06-06:
Here is the line:

'#calc \'wind_chill(data["temp_out"], data["wind_ave"])\' \'"chill": "%.1f", \' #\n'

And it works!

jim-easterbrook, 2019-06-06:
I prefer to use double quotes for the ``#calc`` delimiters and single quotes for the ``dict`` keys, which is the style I've been consistent in using in pywws.
```
#calc "wind_chill(data['temp_out'], data['wind_ave'])" "'chill': '%.1f', "#
```