jim-easterbrook/pywws #88: Upload to Windy.com

mfraser, 2019-08-14:
Would like to be able to upload weather data to windy.com. They have an API at https://community.windy.com/topic/8168/report-you-weather-station-data-to-windy/35

jim-easterbrook, 2019-08-14:
You can create your own "service" module, as described in https://pywws.readthedocs.io/en/latest/guides/integration.html#writing-your-own-uploader
Once you've got it working it'll be easy to add it to the pywws distribution.

closed 2019-11-06 by jim-easterbrook

mfraser, 2019-11-24:
Any idea why this:
pywws.service.windy:server response "SUCCESS (partial) <br> unknown params: 'station_id', <br>"
is showing in the logs?

jim-easterbrook, 2020-01-22:
I've just started testing the Windy uploader (after tweaking it a bit to make it more consistent with other service uploaders) and am getting the same `unknown params: 'station_id'` problem. This doesn't happen when running it from the command line, but does when doing live logging. The station id it's using is from the openweathermap service! Something very strange is happening, possibly related to multi-threading.

jim-easterbrook, 2020-01-22:
Found it - service classes that don't define their own `fixed_data` end up sharing the one defined in the base class!

PR #89 (2019-11-06): service: Add Windy support
Upload data to windy stations https://stations.windy.com/ following the API
defined at https://community.windy.com/post/20776

It does support catchup up to a week from what I've noticed in my tests.

Closes #88
commit 8583f3b3 (2019-11-06): service: Add Windy support

Upload data to windy stations https://stations.windy.com/ following the API
defined at https://community.windy.com/post/20776

It does support catchup up to a week from what I've noticed in my tests.

Closes #88