jim-easterbrook/pywws #90: unorderable types when running pywws-livelog

hrpunio, 2020-02-06:
Debian Stretch/arm when running pywws-livelog. pywws-hourly runs ok
I am running pywws for years w/o problems so looks strange to discover such a 
low-level error
--tomek

07:47:47:pywws.logger:pywws version 20.1.0, build 1673 (92d8868)
07:47:47:pywws.logger:Python version 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516]
07:47:47:pywws.weatherstation:using pywws.device_pyusb1
Traceback (most recent call last):
  File "/usr/local/bin/pywws-livelog", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/pywws/livelog.py", line 130, in main
    return live_log(args[0])
  File "/usr/local/lib/python3.5/dist-packages/pywws/livelog.py", line 75, in live_log
    datalogger = pywws.logdata.DataLogger(context)
  File "/usr/local/lib/python3.5/dist-packages/pywws/logdata.py", line 100, in __init__
    while saved_date < self.last_stored_time:
TypeError: unorderable types: NoneType() < datetime.datetime()

jim-easterbrook, 2020-02-06:
I suspect the `ptr` value in the `[data]` section of your status.ini is corrupted. Delete it and restart.

jim-easterbrook, 2020-02-08:
Commit b571ed6 should reduce the likelihood of this happening in future.

closed 2020-12-04 by jim-easterbrook