jim-easterbrook/pywws #91: error processing hourly

t3amj3ff, 2020-02-09:
Hello

My weather stations been running for years but I've been slack and haven't checked it recently.  Turns out the pi crashed in December but had been giving errors for a few weeks before that.  Hourly processing has stopped working and I can't figure out why.
```
pi@weatherpi2:~/weather $ pywws-hourly -v data
13:04:33:pywws.logger:pywws version 20.1.0, build 1673 (92d8868)
13:04:33:pywws.logger:Python version 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0]
13:04:34:pywws.weatherstation:using pywws.device_libusb1
13:04:35:pywws.logdata:Synchronising to weather station
13:05:00:pywws.weatherstation:status {'lost_connection': False, 'rain_overflow': False}
13:05:00:pywws.logdata:Fetching data
13:05:00:pywws.process:Generating summary data
13:05:01:pywws.calib:Using default calibration
Traceback (most recent call last):
  File "/home/pi/.local/bin/pywws-hourly", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/pywws/hourly.py", line 97, in main
    return hourly(args[0])
  File "/home/pi/.local/lib/python3.7/site-packages/pywws/hourly.py", line 68, in hourly
    pywws.regulartasks.RegularTasks(context).do_tasks()
  File "/home/pi/.local/lib/python3.7/site-packages/pywws/regulartasks.py", line 89, in __init__
    for name, options in self._parse_templates(section, 'services'):
  File "/home/pi/.local/lib/python3.7/site-packages/pywws/regulartasks.py", line 119, in _parse_templates
    for template in literal_eval(self.params.get(section, option, '[]')):
  File "/usr/lib/python3.7/ast.py", line 91, in literal_eval
    return _convert(node_or_string)
  File "/usr/lib/python3.7/ast.py", line 74, in _convert
    return list(map(_convert, node.elts))
  File "/usr/lib/python3.7/ast.py", line 90, in _convert
    return _convert_signed_num(node)
  File "/usr/lib/python3.7/ast.py", line 63, in _convert_signed_num
    return _convert_num(node)
  File "/usr/lib/python3.7/ast.py", line 55, in _convert_num
    raise ValueError('malformed node or string: ' + repr(node))
ValueError: malformed node or string: <_ast.Call object at 0xb4adec10>

```
I thought it might be a corrupt data file but they look ok, I tried removing the one for the day it stopped processing and the one after but no luck.

Any ideas?

Regards

jim-easterbrook, 2020-02-09:
Your weather.ini appears to have errors in one of the 'services' items.

t3amj3ff, 2020-02-09:
Thanks, it was.   The file doesn't look any different but has obviously picked up a corruption somewhere.  I retyped it and it's working again.

closed 2020-02-09 by t3amj3ff