jim-easterbrook/pywws #95: WS3080 not recognised

Nightwalker-87, 2020-08-29:
Hi.
I am facing the same issue as #49 and can reproduce the output (https://github.com/jim-easterbrook/pywws/issues/49#issue-239193494) but with a WS3080.
However I can't find a solution to this currently but am somehow confident that this device will work either.
IMHO the hardware is very similar or even equivalent to the WH3080.
There still seems to be a problem present in the latest version.
Any help appreciated. A workaround would also be helpful.

System: Debian Bullseye (testing)
pywws-version: 20.1.0, build 1673 (92d8868) [installed via debian-repo]
Device-ID: USBDevice(0x1941, 0x8021)
Python versions: 3.8 & 3.9 (2.7 did not seem to have worked either)

ashenshugarRET, 2020-08-30:
Issue #49, seems to have been caused by trying to run pywws on Mac OS X, it was fixed by switching to ubuntu.

Can you confirm which USB library you have installed on your systems, Jim's excellent readthedocs `[documentation](https://pywws.readthedocs.io/en/latest/essentials/dependencies.html#other-systems)` recommends libusb or pyusb before trying hidapi. Have you already tried and failed with libusb and pyusb?

Nightwalker-87, 2020-08-30:
I have both
- libusb-0.1-4 (2020.01.19-1) and
- libusb-1.0-0 (2:1.0.23-2)

installed which are the newest available. I'd prefer to use libusb as well.
I have no clue why pywws tries to use hidapi, that was not intended. The program just came up with it.
The packages
- python3-hid and/or
- python3-hidapi

are not even installed. I haven't tried pyusb yet though.

jim-easterbrook, 2020-08-30:
You'd need to ask the Debian packager for that. The pywws project doesn't provide packages for any particular operating system.

jim-easterbrook, 2020-08-30:
PS pywws was appearing to try and use hidapi as it couldn't find libusb.

Nightwalker-87, 2020-08-30:
> You'd need to ask the Debian packager for that. The pywws project doesn't provide packages for any particular operating system.

I have submitted a request to Debian Maintainer Thorsten Alteholz referring to this ticket.

> PS pywws was appearing to try and use hidapi as it couldn't find libusb.

Can you update the console output for this routine to make visible what is happening in the background, such as
"Detecting pyusb" --> Found / Failed
"Trying to use hidapi instead..." etc.

This would clearly help to improve the initial setup for users and help to detect missing libraries, bindings and modules.

closed 2020-09-16 by jim-easterbrook