jim-easterbrook/pywws #96: logdata.py: TypeError: unsupported operand type(s) for +: 'NoneType' and 'datetime.timedelta'

tinkercnc, 2020-09-30:
Hello,

with fresh pywws installation from git on a raspi I got this error:

 ```
$ pywws-livelog -vvv ~/weather/data

17:35:08:pywws.logger:pywws version 20.9.0, build 1679 (97fbb71)
17:35:08:pywws.logger:Python version 2.7.9 (default, Sep 17 2016, 20:26:04) 
[GCC 4.9.2]
17:35:08:pywws.weatherstation:trying USB module device_libusb1
17:35:08:pywws.weatherstation:trying USB module device_pyusb1
17:35:08:pywws.weatherstation:using pywws.device_pyusb1
'''

17:35:10:pywws.calib:Using default calibration
17:35:10:pywws.service.mqtt:template:
#idx          '"idx"             : "%Y-%m-%d %H:%M:%S",'##wind_dir     '"wind_dir_degrees": "%.d",' '' 'winddir_degrees(x)'##wind_dir     '"wind_dir_text"   : "%s",' '' 'winddir_text(x)'##wind_ave     '"wind_ave_mps"    : "%.2f",'##wind_ave     '"wind_ave_mph"    : "%.2f",' '' 'wind_mph(x)'##wind_gust    '"wind_gust_mps"   : "%.2f",'##wind_gust    '"wind_gust_mph"   : "%.2f",' '' 'wind_mph(x)'##calc 'wind_chill(data["temp_out"],data["wind_ave"])'         '"wind_chill_c" : "%.1f",'##calc 'temp_f(wind_chill(data["temp_out"],data["wind_ave"]))' '"wind_chill_f" : "%.1f",'##calc 'dew_point(data["temp_out"],data["hum_out"])'           '"dew_point_c" : "%.1f",'##calc 'temp_f(dew_point(data["temp_out"],data["hum_out"]))'   '"dew_point_f" : "%.1f",'##hum_out      '"hum_out"     : "%.d",'##hum_in       '"hum_in"      : "%.d",'##temp_in      '"temp_in_c"   : "%.1f",'##temp_in      '"temp_in_f"   : "%.1f",' '' 'temp_f(x)'##temp_out     '"temp_out_c"  : "%.1f",'##temp_out     '"temp_out_f"  : "%.1f",' '' 'temp_f(x)'##calc 'apparent_temp(data["temp_out"],data["hum_out"],data["wind_ave"])'         '"temp_out_realfeel_c" : "%.1f",'##calc 'temp_f(apparent_temp(data["temp_out"],data["hum_out"],data["wind_ave"]))' '"temp_out_realfeel_f" : "%.1f",'##rel_pressure '"pressure_rel_hpa" : "%.1f",'##rel_pressure '"pressure_rel_inhg": "%.4f",' '' 'pressure_inhg(x)'##abs_pressure '"pressure_abs_hpa" : "%.1f",'##abs_pressure '"pressure_abs_inhg": "%.4f",' '' 'pressure_inhg(x)'##rain         '"rain_mm"     : "%.1f",'##rain         '"rain_in"     : "%.2f",' '' 'rain_inch(x)'##calc 'rain_hour(data)'            '"rain_last_hour_mm": "%.1f",'##calc 'rain_inch(rain_hour(data))' '"rain_last_hour_in": "%.2f",'##calc 'rain_24hr(data)'            '"rain_last_24hours_mm": "%.1f",'##calc 'rain_inch(rain_24hr(data))' '"rain_last_24hours_in": "%.2f",'##calc 'rain_day(data)'             '"rain_day_mm": "%.1f",'##calc 'rain_inch(rain_day(data))'  '"rain_day_in": "%.2f",'#
17:35:10:pywws.logdata:Synchronising to weather station
17:35:11:pywws.weatherstation:read period 163
17:35:11:pywws.weatherstation:delay 80, pause 0.5
17:35:11:pywws.weatherstation:status {'rain_overflow': False, 'lost_connection': False}
17:35:12:pywws.weatherstation:delay 80, pause 0.5
17:35:12:pywws.weatherstation:delay 80, pause 0.5
17:35:13:pywws.weatherstation:delay 80, pause 0.5
17:35:14:pywws.weatherstation:delay 80, pause 0.5
17:35:14:pywws.weatherstation:delay 80, pause 0.5
17:35:15:pywws.weatherstation:delay 80, pause 0.5
17:35:16:pywws.weatherstation:delay 80, pause 0.5
17:35:16:pywws.weatherstation:delay 80, pause 0.5
17:35:17:pywws.weatherstation:delay 80, pause 0.5
17:35:18:pywws.weatherstation:delay 80, pause 0.5
17:35:18:pywws.weatherstation:delay 80, pause 0.5
17:35:19:pywws.weatherstation:live_data new data
17:35:19:pywws.weatherstation:setting sensor clock 7.68365
17:35:19:pywws.logdata:Reading time 15:35:19
17:35:19:pywws.logdata:est log time 14:13:55 +- 24s (14:13:31..14:14:19)
17:35:19:pywws.weatherstation:delay 81, pause 44.4828
17:36:04:pywws.weatherstation:avoid 5.89973092079
17:36:11:pywws.weatherstation:live_data new data
17:36:11:pywws.logdata:Reading time 15:36:07
17:36:11:pywws.logdata:est log time 14:14:13 +- 6s (14:14:07..14:14:19)
17:36:11:pywws.logdata:Fetching data
17:36:11:pywws.livelog:unsupported operand type(s) for +: 'NoneType' and 'datetime.timedelta'
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pywws-20.9.0-py2.7.egg/pywws/livelog.py", line 80, in live_log
    datalogger.log_data()
  File "/usr/local/lib/python2.7/dist-packages/pywws-20.9.0-py2.7.egg/pywws/logdata.py", line 238, in log_data
    self.fetch_logged(last_date, last_ptr)
  File "/usr/local/lib/python2.7/dist-packages/pywws-20.9.0-py2.7.egg/pywws/logdata.py", line 173, in fetch_logged
    next_date = self.raw_data.after(last_date + SECOND)
TypeError: unsupported operand type(s) for +: 'NoneType' and 'datetime.timedelta'
17:36:11:pywws.storage:flushing

```
thanks in advance

jim-easterbrook, 2020-09-30:
First point to note is your station has a "logging interval" of 163 minutes, so will only be storing data at nearly 3 hour intervals. This is unlikely to work well with pywws, which is usually used with a period of 5 or 10 minutes.

The unsupported operand problem can be caused by a corrupt `status.ini` file. This file can safely be deleted (while pywws is stopped).

tinkercnc, 2020-09-30:
Thank you very much! Now it works. :)

closed 2020-09-30 by tinkercnc

jim-easterbrook, 2020-10-01:
Looking more closely I think the problem can occur if there is no stored data. Commit 9b070a5 should fix it, whatever the cause.