jim-easterbrook/pywws #98: Problem to run weatherstation.py "No USB libary found"

Hoggel0815, 2020-11-11:
Hi,

i try to run on my RaspberryPi4 with Python3.7 the weatherstation.py, to get some easy weatherinformations for a own python Projekt, but everything i´ve try to run the programm doesent work and i end up with No USB libary found".

I´ve trying "pip3 install libusb", "pip3 install usb", "sudo apt-get install python-libusb1.0.0", "sudo apt-get install python-usb", "pip3 install pyusb"

Any ideas to try something else?

jim-easterbrook, 2020-11-11:
Try `sudo apt-get install python3-libusb1` to install the Python 3 version instead of the Python 2 version.

Hoggel0815, 2020-11-11:
Thanks for the fast answere, but the problem still "No USB libary found".
I try to run the example code in weatherstation.py

`import weatherstation

ws = weatherstation.WeatherStation()
print(ws.get_fixed_block(['min', 'temp_out', 'val']))`

jim-easterbrook, 2020-11-11:
Remove all the pip installed USB stuff and reinstall python3-libusb1 with apt. As it relies on a C library (libusb1) it is much safer to install with apt as that should install the C dependency. then try running pywws-testweatherstation.

Hoggel0815, 2020-11-11:
I´ve tried it on a fresh other RaspberryPi to run the testweatherstation, but the same issue

jim-easterbrook, 2020-11-11:
What's the output of `dpkg -l | grep -i libusb`?

Hoggel0815, 2020-11-11:
ii  libusb-0.1-4:armhf                    2:0.1.12-32                            armhf        userspace USB programming library
ii  libusb-1.0-0:armhf                    2:1.0.22-2                             armhf        userspace USB programming library
ii  libusb-1.0-doc                        2:1.0.22-2                             all          documentation for userspace USB programming
ii  libusbmuxd4:armhf                     1.1.0~git20181007.07a493a-1            armhf        USB multiplexor daemon for iPhone and iPod Touch devices - library
ii  python3-libusb1                       1.7-1                                  all          Python wrapper for libusb1 (Python 3)

jim-easterbrook, 2020-11-11:
That's all looking good. What about `python3 -c "import pywws"`?

Hoggel0815, 2020-11-11:
that work´s no output

jim-easterbrook, 2020-11-11:
OK, I think `python3 -m pywws.testweatherstation` should work, in which case pywws is installed and working.

Hoggel0815, 2020-11-11:
Yeah, but

Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/.local/lib/python3.7/site-packages/pywws/testweatherstation.py", line 190, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/pywws/testweatherstation.py", line 117, in main
    ws = pywws.weatherstation.WeatherStation()
  File "/home/pi/.local/lib/python3.7/site-packages/pywws/weatherstation.py", line 464, in __init__
    self.cusb = CUSBDrive()
  File "/home/pi/.local/lib/python3.7/site-packages/pywws/weatherstation.py", line 318, in __init__
    self.dev = USBDevice(0x1941, 0x8021)
  File "/home/pi/.local/lib/python3.7/site-packages/pywws/device_libusb1.py", line 77, in __init__
    self.dev = self.context.openByVendorIDAndProductID(idVendor, idProduct)
  File "/usr/lib/python3/dist-packages/usb1/__init__.py", line 2391, in openByVendorIDAndProductID
    return result.open()
  File "/usr/lib/python3/dist-packages/usb1/__init__.py", line 2159, in open
    mayRaiseUSBError(libusb1.libusb_open(self.device_p, byref(handle)))
  File "/usr/lib/python3/dist-packages/usb1/__init__.py", line 133, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/usr/lib/python3/dist-packages/usb1/__init__.py", line 125, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorAccess: LIBUSB_ERROR_ACCESS [-3]

jim-easterbrook, 2020-11-11:
That's probably because your user account doesn't have permission to access the station's USB port. You can either run everything as root (not a good idea) or set a udev rule to change ownership or permissions on the USB port.
https://pywws.readthedocs.io/en/latest/guides/getstarted.html?highlight=udev#test-the-weather-station-connection

Hoggel0815, 2020-11-11:
Ok, my Mission was to get simple weather informations like temperature in and out, wind speed, wind direction, rainfall.

So i want to create a object from weatherherstation and thought to get informations over ws.get_fixed_block(['min', 'temp_out', 'val'])` in my programm, for example.

Is that feasible?

jim-easterbrook, 2020-11-11:
The "fixed block" probably isn't what you want - it stores things like the highest and lowest temperature the station has ever recorded. Have a look at the pywws.testweatherstation module. Its -h option fetches the most recently logged data which includes temperatures and wind direction.

Rainfall is slightly tricky as the data is a cumulative amount since the outside sensors were last reset. You'll need to store a previous reading and compare the two to see how much rain has fallen in that time interval.

Hoggel0815, 2020-11-12:
import pywws.weatherstation
import pprint

def raw_dump(pos, data):
    print("%04x" % pos, end=' ')
    for item in data:
        print("%02x" % item, end=' ')
    print('')

ws = pywws.weatherstation.WeatherStation()

history_count = 10

fixed_block = ws.get_fixed_block()
ptr = fixed_block['current_pos']
for i in range(history_count):
	print("0x%04x" % ptr, end=' ')
	data = ws.get_data(ptr)
	pprint.pprint(data)
	ptr = ws.dec_ptr(ptr)

Now i got enough to work with, you are great thanks a lot @jim-easterbrook

closed 2020-11-12 by Hoggel0815