jim-easterbrook/pywws #99: Can't change station read period

bertieuk, 2021-03-05:
Running _pywws-setweatherstation -r 5_ doesn't change my station read period and am stuck on 30 mins.

`# pywws-setweatherstation -vvv -r 5
10:46:12:pywws.logger:pywws version 20.1.0, build 1673 (92d8868)
10:46:12:pywws.logger:Python version 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0]
10:46:12:pywws.weatherstation:using pywws.device_pyusb1
10:46:13:pywws.weatherstation:write_data waiting for ack
10:46:19:pywws.weatherstation:write_data waiting for ack
10:46:25:pywws.weatherstation:write_data waiting for ack
10:46:31:pywws.weatherstation:write_data waiting for ack`

`# pywws-testweatherstation -vvv -d
10:46:46:pywws.logger:pywws version 20.1.0, build 1673 (92d8868)
10:46:46:pywws.logger:Python version 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0]
10:46:46:pywws.weatherstation:using pywws.device_pyusb1
{'abs_pressure': 965,
.
.
 'current_pos': 800,
 'data_changed': 0,
 'data_count': 35,
 'date_time': '2021-03-05 10:45:00',
.
.
.
 'rain_factor_raw': 17920,
 'read_period': 30,
.
.
 'timezone': 0,
 'unknown_01': 0,
 'unknown_18': 0,
 'wind_factor_raw': 91}
min -> temp_out -> {'val': -36.7, 'date': '2013-10-02 07:29:00'}
alarm -> hum_out -> {'hi': 70, 'lo': 45}
rel_pressure -> 1019.3
abs_pressure -> 965`

Maplin W1080 connected to a Pi 3B, clean install of Raspbian Lite + dependancies. 

Tried: 
python 2.7 v python 3.7
libusb1 v pyusb
multiple different power supplies
Pi 3B+
/boot/config.txt dtoverlay settings from another suggestion
Unplugging W1080 & removing batteries for over an hour

Nothing works and read_period in test output always stays at 30. Any suggestions?

jim-easterbrook, 2021-03-05:
No idea, sorry. It sounds as if your weather station is defective. Can you try it with "EasyWeather" on Windows?

bertieuk, 2021-03-05:
> No idea, sorry. It sounds as if your weather station is defective. Can you try it with "EasyWeather" on Windows?

Tried EasyWeather 9.3 on Windows XP VirtualBox, can see the station but get 'Failure' when attempting to save the read period change.

Looks like it's time for a new one :-(

closed 2024-11-27 by jim-easterbrook