jim-easterbrook/pywws #100: WH1080 suddenly fires alarms

mrumpf, 2021-03-13:
I have a WH1080 whether station that suddenly started firing alarms (loud beeps in the middle of the night) for all the measured values (pressure, temperature, humidity, ...). The only thing I did recently was to update pywws so I wonder whether this update could have triggered the alarm settings. It was running pywws for 3 years now without alarm settings at all. I thought before I destroy the piezospeaker on the display unit I ask whether this might be caused by a recent update and whether this can be controlled by the pywws library to avoid any physical force against the device :)

jim-easterbrook, 2021-03-13:
I don't believe anything in pywws can cause this. (But feel free to study the GitHub commits to find a relevant change between your previous and current version.)

My station has several faults, one of which is to set alarms (and change the display units) when I do a power off reset. Which is why the latest `pywws-setweatherstation` has an option to set sensible defaults for most of the options.

mrumpf, 2021-03-13:
Fortunately I have an older output of "pywws-testweatherstation -d" where the alarm1-3 sections are all set to "False":
````
pywws-testweatherstation -d
21:51:20:pywws.logger:pywws version 18.10.1, build 1646 (50de9af)
{'abs_pressure': 888.3,
 'alarm': {'abs_pressure': {'hi': 1040, 'lo': 960},
           'dewpoint': {'hi': 10, 'lo': -10},
           'hum_in': {'hi': 65, 'lo': 35},
           'hum_out': {'hi': 70, 'lo': 45},
           'illuminance': 0,
           'rain': {'day': 150, 'hour': 3},
           'rel_pressure': {'hi': 1040, 'lo': 960},
           'temp_in': {'hi': 20, 'lo': 0},
           'temp_out': {'hi': 30, 'lo': -10},
           'time': '12:00',
           'uv': 0,
           'wind_ave': {'bft': 0, 'ms': 18},
           'wind_dir': 0,
           'wind_gust': {'bft': 0, 'ms': 10.4},
           'windchill': {'hi': 20, 'lo': 0}},
 'alarm_1': {'bit0': False,
             'bit3': False,
             'hum_in_hi': False,
             'hum_in_lo': False,
             'hum_out_hi': False,
             'hum_out_lo': False,
             'time': False,
             'wind_dir': False},
 'alarm_2': {'pressure_abs_hi': False,
             'pressure_abs_lo': False,
             'pressure_rel_hi': False,
             'pressure_rel_lo': False,
             'rain_day': False,
             'rain_hour': False,
             'wind_ave': False,
             'wind_gust': False},
 'alarm_3': {'dew_point_hi': False,
             'dew_point_lo': False,
             'temp_in_hi': False,
             'temp_in_lo': False,
             'temp_out_hi': False,
             'temp_out_lo': False,
             'wind_chill_hi': False,
             'wind_chill_lo': False},
...
````

I updated the OS recently but it seems as if the version was not changed by that update. Debian Buster still has the 18.10.1 version. I remove the package with "apt remove --purge python3-pywws" and installed the latest version using "pip3 install pywws"

The most recent output of the "pywws-testweatherstation -d" shows that several Bits in alarm_1 - alarm_3 have been set to True now.
````
$ pywws-testweatherstation -d
12:15:38:pywws.logger:pywws version 20.1.0, build 1673 (92d8868)
{'abs_pressure': 928.1,
 'alarm': {'abs_pressure': {'hi': 1040, 'lo': 960},
           'dewpoint': {'hi': -12.2, 'lo': -23.3},
           'hum_in': {'hi': 65, 'lo': 35},
           'hum_out': {'hi': 70, 'lo': 45},
           'illuminance': 0,
           'rain': {'day': 150, 'hour': 3},
           'rel_pressure': {'hi': 1040, 'lo': 960},
           'temp_in': {'hi': 20, 'lo': 0},
           'temp_out': {'hi': -1.1, 'lo': -23.3},
           'time': '12:00',
           'uv': 0,
           'wind_ave': {'bft': 0, 'ms': 18},
           'wind_dir': 0,
           'wind_gust': {'bft': 0, 'ms': 10.4},
           'windchill': {'hi': -6.7, 'lo': -17.8}},
 'alarm_1': {'bit0': False,
             'bit3': True,
             'hum_in_hi': False,
             'hum_in_lo': False,
             'hum_out_hi': False,
             'hum_out_lo': False,
             'time': False,
             'wind_dir': False},
 'alarm_2': {'pressure_abs_hi': False,
             'pressure_abs_lo': False,
             'pressure_rel_hi': True,
             'pressure_rel_lo': True,
             'rain_day': False,
             'rain_hour': True,
             'wind_ave': False,
             'wind_gust': False},
 'alarm_3': {'dew_point_hi': True,
             'dew_point_lo': True,
             'temp_in_hi': False,
             'temp_in_lo': False,
             'temp_out_hi': True,
             'temp_out_lo': True,
             'wind_chill_hi': False,
             'wind_chill_lo': False},
...
````

mrumpf, 2021-03-13:
When I read the code correctly, the "pywws-setweatherstation -d" should write 0 to the alarm section, turning off all alarms:
````
if default:
        ptr = ws.fixed_format['settings_1'][0]
        data.append((ptr, 0b00100000))
        ptr = ws.fixed_format['settings_2'][0]
        data.append((ptr, 0b00001000))
        ptr = ws.fixed_format['display_1'][0]
        data.append((ptr, 0b01000001))
        ptr = ws.fixed_format['display_2'][0]
        data.append((ptr, 0b00001001))
        ptr = ws.fixed_format['alarm_1'][0]
        data.append((ptr, 0b00000000))
        ptr = ws.fixed_format['alarm_2'][0]
        data.append((ptr, 0b00000000))
        ptr = ws.fixed_format['alarm_3'][0]
        data.append((ptr, 0b00000000))
````
But it seems as if this code Is not yet included in version 20.1.0.

jim-easterbrook, 2021-03-13:
Ah, you're right - it was added in June 2020. Looks like a new release is overdue.

closed 2021-03-13 by mrumpf

jim-easterbrook, 2021-03-15:
I've just released version 21.3.0 with this included.