jim-easterbrook/pywws #101: version 21.3.0 is crashing after clocks change

ashenshugarRET, 2021-03-28:
Version 21.3.0 is crashing with the following error post the clock change:

```
2021-03-28 08:07:43:pywws.regulartasks:Templating 24hrs_wind_rose_highchart.csv
2021-03-28 08:07:44:pywws.livelog:2021-03-28 01:01:31
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pywws/livelog.py", line 89, in live_log
    tasks.do_tasks()
  File "/usr/local/lib/python3.7/dist-packages/pywws/regulartasks.py", line 221, in do_tasks
    self._do_common(now, sections)
  File "/usr/local/lib/python3.7/dist-packages/pywws/regulartasks.py", line 149, in _do_common
    self.services[name].upload(live_data=live_data, options=options)
  File "/usr/local/lib/python3.7/dist-packages/pywws/service/__init__.py", line 340, in upload
    if self.queue_data(timestamp, data):
  File "/usr/local/lib/python3.7/dist-packages/pywws/service/__init__.py", line 307, in queue_data
    OK = super(CatchupDataService, self).queue_data(timestamp, data)
  File "/usr/local/lib/python3.7/dist-packages/pywws/service/__init__.py", line 281, in queue_data
    prepared_data = self.prepare_data(data)
  File "/usr/local/lib/python3.7/dist-packages/pywws/service/__init__.py", line 290, in prepare_data
    data_str = self.templater.make_text(self.template_file, data)
  File "/usr/local/lib/python3.7/dist-packages/pywws/template.py", line 578, in make_text
    for text in self.process(live_data, template_file):
  File "/usr/local/lib/python3.7/dist-packages/pywws/template.py", line 460, in process
    x = eval(command[1])
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/pywws/template.py", line 343, in rain_day
    midnight = timezone.local_midnight(data['idx'])
  File "/usr/local/lib/python3.7/dist-packages/pywws/timezone.py", line 131, in local_midnight
    return self.local_replace(dt, hour=0, minute=0, second= 0)
  File "/usr/local/lib/python3.7/dist-packages/pywws/timezone.py", line 107, in local_replace
    dst_offset = self.dst(local_time)
  File "/usr/local/lib/python3.7/dist-packages/pywws/timezone.py", line 68, in dst
    return self.local.dst(dt)
  File "/usr/local/lib/python3.7/dist-packages/pytz/tzinfo.py", line 461, in dst
    dt = self.localize(dt, is_dst)
  File "/usr/local/lib/python3.7/dist-packages/pytz/tzinfo.py", line 341, in localize
    raise NonExistentTimeError(dt)
pytz.exceptions.NonExistentTimeError: 2021-03-28 01:01:31
```

Rolling back to the previous version with:

`sudo pip3 install pywws==20.1.0`

Fixes the issue so the problem is with the most recent update.

Thanks

jim-easterbrook, 2021-03-28:
What version of pytz are you using?

ashenshugarRET, 2021-03-28:
I was using version 2019.1 and updated to version 2021.1

```
richard@WeatherStation:~$ sudo pip install --upgrade pytz
Requirement already satisfied: pytz in /usr/local/lib/python3.7/dist-packages (2019.1
Collecting pytz
  Downloading pytz-2021.1-py2.py3-none-any.whl (510 kB)
     |████████████████████████████████| 510 kB 2.6 MB/s 
Installing collected packages: pytz
  Attempting uninstall: pytz
    Found existing installation: pytz 2019.1
    Uninstalling pytz-2019.1:
      Successfully uninstalled pytz-2019.1
Successfully installed pytz-2021.1
```
Thanks

jim-easterbrook, 2021-03-28:
Does pywws v 21.3.0 work after updating pytz? (You can run 'python3 -m pywws.timezone' to test it.)

ashenshugarRET, 2021-03-28:
Yes, it works:

```
richard@WeatherStation:/usr/local/lib/python3.7/dist-packages/pywws$ python3 -m pywws.timezone
using timezone Europe/London
current time
2021-03-28 09:30:45 UTC
2021-03-28 10:30:45+01:00 BST
2021-03-28 09:30:45+00:00 UTC

most recent 9am 2021-03-28 08:00:00 UTC
most recent 9pm 2021-03-27 21:00:00 UTC

Ambiguous / missing times in "Europe/London"
2020-03-28 23:15:00 UTC = 2020-03-28 23:15:00+00:00 GMT
2020-03-29 00:15:00 UTC = 2020-03-29 00:15:00+00:00 GMT
2020-03-29 01:15:00 UTC = 2020-03-29 02:15:00+01:00 BST
2020-03-29 02:15:00 UTC = 2020-03-29 03:15:00+01:00 BST

2020-10-24 23:15:00 UTC = 2020-10-25 00:15:00+01:00 BST
2020-10-25 00:15:00 UTC = 2020-10-25 01:15:00+01:00 BST
2020-10-25 01:15:00 UTC = 2020-10-25 01:15:00+00:00 GMT
2020-10-25 02:15:00 UTC = 2020-10-25 02:15:00+00:00 GMT

DST transitions in "America/St_Johns"
midnight 2018-03-11 03:30:00 UTC 2018-03-11 00:00:00-03:30 NST
midnight 2018-03-12 02:30:00 UTC 2018-03-12 00:00:00-02:30 NDT

midnight 2018-11-04 02:30:00 UTC 2018-11-04 00:00:00-02:30 NDT
midnight 2018-11-05 03:30:00 UTC 2018-11-05 00:00:00-03:30 NST
```

closed 2021-03-28 by ashenshugarRET

jim-easterbrook, 2021-03-28:
Thanks, now I can narrow down the minimum version of pytz needed and then modify pywws accordingly.