jim-easterbrook/pywws #103: 70.8mm of rain, turned into 7.8mm in pywws

Git2263, 2021-05-31:
Pywws download 01.04.2021
The weather panel shows 70.8mm of rain, data from pywws shows only 7.8mm.

jim-easterbrook, 2021-05-31:
Which pywws data file? (raw, hourly, ...)
What period is the station console showing? (hour, 24 hour, ...)

Git2263, 2021-05-31:
Data hourly:
2021-05-30 18:57:59,31,25.3,45,15.4,989.7,1023.3,-0.2,1.33333333333,4.1,8.54526356906,0
2021-05-30 19:57:59,32,24.9,41,19.7,989.8,1023.4,0.1,0.791666666667,2,15.3363814587,7.8
2021-05-30 20:57:59,33,24.2,59,16.2,990,1023.6,0.4,1,2.4,11.7830204612,0

Panel hourly 
70.8

The water comes from a bottle I used for test.

jim-easterbrook, 2021-05-31:
Rain rates of more than 5 mm per minute are assumed to be corrupt data and are ignored. (There can be problems if the wind gets into your rain gauge and rattles the see-saw from side to side.) You should get a warning message, depending on the verbosity you run pywws at.

If you have exceptionally high rainfall rates where you live then you can edit `pywws/process.py` and change line 266 accordingly.

Git2263, 2021-05-31:
Okay, there are many places where it rains a lot in a short time, so you might want to comment this in a text file :)

jim-easterbrook, 2021-05-31:
These cheap weather stations won't accurately record such high rainfall. Most of it will bounce off the funnel.