jim-easterbrook/pywws #114: No 'rain' processing should occur if old status is 'lost connection'

diceit-git, 2023-12-30:
After reconnecting the raw data look like:

```
2023-11-25 12:56:20,30,61,9.5,83,1,942.5,0.7,1.4,14,243.6,0,7943.2,0
2023-11-25 13:00:20,4,61,9.2,,,942.7,,,,0,64,,
...
2023-11-25 14:56:33,30,64,8.8,,,942.2,,,,0,64,,
2023-11-25 15:26:28,30,64,8.8,86,1.1,942.3,0.7,1.4,4,243.6,0,,
2023-11-25 15:56:02,30,65,9,87,1,942.3,0.3,0.7,8,243.6,0,,
```
process.py calculates the rain delta without recognizing, that the WSStatus of previous value is "not connected", so you get a peek in rain values of, in this case , 243.6mm, which is incorrect.

The WSStatus should be recognized and rain value only be calculated if both readings are valid

jim-easterbrook, 2023-12-30:
That makes sense, but maybe with a time comparison so that short bursts of lost connection don't cause too much of a problem.

I get occasional lost connection and have never noticed this problem though. But my logging interval is the recommended 5 minutes rather than the 30 minutes you are using.

diceit-git, 2023-12-30:
Hi Jim,
maybe it's working 'accidentally' with 5 min, because the value exceeds the delay*5 border, if I look at the code.