jim-easterbrook/pywws #115: Twitter Posting

rick165, 2024-02-08:
Is there any fix to the following error since twitter changed the api access?
:pywws.service.twitter:TwitterError([{u'message': u'You currently have access to a subset of Twitter API v2 endpoints and limited v1.1 endpoints (e.g. media post, oauth) only. If you need access to this endpoint, you may need a different access level. You can learn more here: https://developer.twitter.com/en/portal/product', u'code': 453}],)

jim-easterbrook, 2024-02-08:
No, unless someone wants to pay Twitter $100 per month for pywws to be able to post messages.

rick165, 2024-02-08:
I did wonder after it stopped working in July 23 and it hadn't been fixed yet. Shame really, oh well I'll post the stats on mastodon instead. Thanks for the feedback Jim, been using pywws for longer than I can remember 😊

closed 2024-02-08 by rick165

jim-easterbrook, 2024-02-19:
pywws has more than one user. If you create your own developer account and generate your own key/secret then you can post to Twitter. But if you share that key/secret (as pywws does) then all the users soon add up to more than 1000 posts per month.

rick165, 2024-02-19:
Apologies, I'd always used pywws with my own twitter account in isolation.