jim-easterbrook/pywws #122: BlueSky session not working

jaredearle, 2025-01-07:
I can't get the BlueSky service to work. I keep getting this error

`-> % python -m pywws.service.bluesky ~/weather/data skeet.txt`          
`00:31:39:pywws.logger:pywws version 24.8.0, build 1710 (803f8e5)`
`00:31:41:__main__:cannot unpack non-iterable Session object`
`Traceback (most recent call last):`
`  File "/usr/local/lib/python3.9/dist-packages/pywws/service/__init__.py", line 180, in run`
`    OK = self.upload_batch()`
`  File "/usr/local/lib/python3.9/dist-packages/pywws/service/__init__.py", line 443, in upload_batch`
`    with self.session() as (session, message):`
`TypeError: cannot unpack non-iterable Session object`

jim-easterbrook, 2025-01-07:
This appears to be an error in PR #121. In fact I'm surprised it works at all. The `session` method should yield a tuple of `(session, status message)`.

The status message allows errors in trying to establish a session to be reported.

jaredearle, 2025-01-07:
I have tried this with an app password and the user's main password, so we can rule that out.